onlaj/Piano-LED-Visualizer #362: Connecting PC to RPI without Piano

samin50, 2022-06-17:
Hi,
My understanding is that the RPI is configured as Host and can't switch dynamically (you can change it, but I cannot do this for other reasons) between Host and Device/Agent, but I'm wondering if it is possible to configure my PC as a Device to output MIDI to the visualizer which will then light up the corresponding LEDs, without the MIDI-MIDI USB interface talked about in other posts?

My reasoning for doing this is that I plan to output my piano MIDI to my PC which will capture it, send it to a visualiser program I have, and then route it as an output back to the RPI to then light the LEDs as I play the notes using a routing software like MIDI-OX.

When connecting the RPI to my PC, there is no option to set my PC as input, which is why I want to try and configure a USB port as 'emulating' an actual MIDI port for the RPI to pick up, or alternatively as a MIDI Device output.

Is this possible?

Thank you.

ozturkkl, 2022-06-18:
Yes, this is possible, I have a similar setup. Although the connection is not wired, it's wireless.

You can follow the RTPMidi instructions to achieve this. You have to install RTPMidi on your PC which allows you to send MIDI data over the network to the Pi. I also am using the piano midi output to control my DAW and other VSTs so using RTP midi was the only option for me. 

https://github.com/onlaj/Piano-LED-Visualizer/blob/master/Docs/manual_installation.md#6-installing-rtp-midi-server-optional
https://github.com/onlaj/Piano-LED-Visualizer#learning-to-play-with-synthesia

RTPMidi creates a virtual midi session that the pi visualizer can connect over the network. Then the only thing left to do is to tell your DAW to send midi data to the RTPMidi virtual port.

sreich, 2022-07-21:
Perfect thank you this is exactly what I was looking for. It didn't even occur to me that wireless would be an easy option, that's awesome. One less wire routing through my house

sreich, 2022-07-27:
@ozturkkl what'd you use for a diffuser?

closed 2022-09-10 by samin50