onlaj/Piano-LED-Visualizer #371: LCD Stays White

ausfalt, 2022-08-08:
I went for the quick setup via Etcher imager and got the LEDs to work, controlling using the web interface. However, I also have a waveshare 1,44 LCD hat connected and it just stays white. I was able to enable/disable the display using the web interface but all it shows is a blank white screen.

MilanScherpen, 2022-08-18:
Same problem here......

onlaj, 2022-08-18:
Hey.
Try to run https://github.com/onlaj/Piano-LED-Visualizer/blob/master/tests/test_screen.py to see if it works with demo code.
I had this problem once, my screen suddenly became white and stayed like that for ~3 days. No idea what caused that and what fixed the problem, I just left it running.

MilanScherpen, 2022-08-18:
@onlaj thanks for your quick response. As soon as i configured the internet connection the device worked. Thanks any way🙂

MilanScherpen, 2022-08-22:
> Hey. Try to run https://github.com/onlaj/Piano-LED-Visualizer/blob/master/tests/test_screen.py to see if it works with demo code. I had this problem once, my screen suddenly became white and stayed like that for ~3 days. No idea what caused that and what fixed the problem, I just left it running.

Apparently the device only works with the internet, while I don't think it should. It doesn't work for me without WiFi, but as soon as it's connected again it does. Any idea what could be causing this or how I might be able to fix this?

onlaj, 2022-08-22:
Try to add `--webinterface false` to starting parameters

`sudo python3 /home/Piano-LED-Visualizer/visualizer.py --webinterface false`

closed 2022-11-02 by onlaj