onlaj/Piano-LED-Visualizer #379: List of midifiles not updated

AL1-1956, 2022-08-31:
Hello 
Maybe i'm doing something wrong or have'nt read in the doc but I have one problem
When I upload new midi files in the Pi by mean of the Web Interface , I can see in SSH that files are updated
but when I go into the "load song" menu on the LCD , it is still the old list

One example 
I upload 4 files named 046.mid 047.mid 048.mid 049.mid
then I reboot the PI

pi@raspberrypi:/home/Piano-LED-Visualizer/Songs $ ls
046.mid  047.mid  048.mid  049.mid  cache

the files are in the load song menu of the LCD

then I erase these files (Trashcan Web Interface) and upload 4 new files 050.mid 051.mid 052.mid 053.mid
 
pi@raspberrypi:/home/Piano-LED-Visualizer/Songs $ ls
050.mid  051.mid  052.mid  053.mid  cache

So the files are indeed in the Pi memory , but when I try to load them from the LCD menu "load song" , it is still the old files that i see

To see the new files I have to reboot the PI 

Is this normal ??

Have a nice day

onlaj, 2022-08-31:
> So the files are indeed in the Pi memory , but when I try to load them from the LCD menu "load song" , it is still the old files that I see

What about songs list on web interface? Does it update?
My RPi is damaged so currently I'm not able to test it.

AL1-1956, 2022-08-31:
Hello and thanks for the quick answer
The files are updated in the web interface and in the PI but the Visualizer still shows the old list of files wich do not exist anymore .
The only way for me to see the new files in the Visualizer is to rebot the PI.

onlaj, 2022-08-31:
After uploading new files go to menu -> Ports Settings -> Refresh ports

It should also refresh songs list.

AL1-1956, 2022-08-31:
Okay .
So I did the "Refresh ports" trick and it worked.
No need to reboot the PI after a change of the playlist

And for my other problem (Issue #380) about raspi-config not working , forget it , it is now OK too !!
I still have one question , since I do not need the rtpmidi and the autoconnect how can I disable them ??

Thank you so much for all the backup and help you give on this project.

Alain

closed 2022-11-02 by onlaj