onlaj/Piano-LED-Visualizer #381: No more PI user by default on new version of "Raspberry Pi OS Bullseye"

AL1-1956, 2022-09-01:
Hello.
I tried to make a manual installation of PLV 1.4 on a fresh SD Card and could never connect by SSH

the trick with the empty ssh file and the wpa_supplicant.conf file do not work anymore in last version of OS

https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/

So the guide for manual installation needs an update.

Kindly
Alain

francescodicosmo, 2023-11-06:
Hello,
I am having the same problem with PLV 1.5.1.

Ferogreywolf7, 2023-11-11:
Could you not use "sudo adduser pi" to make a new user?

PeterBouSaada, 2024-06-04:
Enabling SSH and adding a wifi configuration is possible with a headless configuration using [Raspberry Pi Imager](https://www.raspberrypi.com/software/).