onlaj/Piano-LED-Visualizer #382: LED Color Accuracy

ausfalt, 2022-09-05:
Not too familiar with LED lights but I wanna check if this has something to do with the LED strip or otherwise?

I'm trying to do a yellow color (#FFFF00)  but on the LED, it looking greenish (yellow-green). Same observation from my friends. 

I have attached a screenshot from the web interface:
[image]

onlaj, 2022-09-05:
Some leds are going into greenish indeed.

[image]

(i swear it's not as dusty as it looks on photo :D)

In my case it was always like this, each led has it's own hue. Maybe more expensive and higher quality led strips are immune to that.
Try to fiddle around with different color values.

ausfalt, 2022-09-05:
Thanks man. Definitely will adjust based on the actual light instead of the specific color input then. 

Cheers.

closed 2022-11-02 by onlaj