onlaj/Piano-LED-Visualizer #383: [bug] holding 2 keys adjacent keys and releasing turns off leds

sreich, 2022-09-07:
This is a bug I noticed. I may end up fixing this myself if I can find time

When you hold two adjacent keys conf down And you lift one of those up, if the option to have the side LEDs lit up is enabled, it will actually turn off the LEDs of the other key

So basically on my setup I have it where each key lights up three LEDs. This means the lights kind of intersect which is fine except when they are released it will turn off too many of the LEDs

I suppose we should check if the side LED is supposed to be lit up. I'm not sure how this is stored yet and if it's information it already has or if this is a more invasive change. I suspect it shouldn't be too bad

onlaj, 2022-10-21:
https://github.com/onlaj/Piano-LED-Visualizer/commit/3bf1a55ecd0e516b57d6bf1b8d56bc6aeab7437f

Hey, thanks for reporting. Quick fix for now, doesn't work properly with Velocity mode. Will be fixed along https://github.com/onlaj/Piano-LED-Visualizer/issues/390 issue

closed 2022-11-02 by onlaj

sreich, 2022-11-02:
Thank you for taking the time to fix!