onlaj/Piano-LED-Visualizer #388: I can't figure out the wiring

Rodrix33333, 2022-10-07:
I just burned a Raspberry Pi Zero of 70 dollars.
I have just one left and I cannot get a replacement in my country at any price. It's just not available.

Please help so I don't mess up this time.

Did I get the correct pins?
Did I solder them correctly? (First time)

Raspberry Pi booted and showes menu. When I connectes to the led strip (which I double checked with another controller that works) the led strip turned off.

No lights.

So I started switching the cables until I fried it. Please help me find my mistake. It's my last chance.

Even with enough money I will have to wait 60 shipping days to get a new Raspberry Pi Zero if I fail...

I would appreciate help.

I need a photo of the reverse of the Raspberry Pi and how to solder. I need to know what do with the remaining res cable of the LED strip.

[image]
[image]

onlaj, 2022-10-07:
Hey, you need to provide more detailed photos, with whole wiring visible.

Rodrix33333, 2022-10-07:
Wow. Thanks for the super fast reply.

So my led strip was made for a nodemcu esp8266, with WLED.

The Nodemcu controller was attached to 3 wires. Red (5v), black (ground), and green (data).

I disconnectes the Nodemcu and wired the black and green into the Raspberry Pi. Not using the red.

Was this correct?

I can get you more photos as soon as I enter home.
Thanks a million in advance 💗

onlaj, 2022-10-07:
Well, first off you also need power supply. Check the schematics here: https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/

Rodrix33333, 2022-10-07:
Yes, I have a power supply for the Raspberry Pi.
And I also have a power supply for the Leds.

The image of the Raspberry Pi of that page soe snot match my Raspberry Pi Zero. So I don't understand it.

Could you provide me a photo of the back of the Raspberry Pi and it's soldering? Thanks in advance.

Is mine correct?
[image]

onlaj, 2022-10-07:
> The image of the Raspberry Pi of that page does not match my Raspberry Pi Zero. So I don't understand it.

It does, as seen from top. Here is my photo: https://imgur.com/7KhwM7r

Your picture doesn't load for me.

Rodrix33333, 2022-10-08:
I just arrived home and took some photos and uploaded them.

This is my setup:
https://postimg.cc/MMYLb0J3

The led controller has its own power supply and ends with two cables. One for the power supply, the other to generic led controller made with Nodemcu.

I unplugged the Nodemcu and plugged new cables soldered to the Raspberry Pi, soldering white and green and not using the red one.

Photo with generic controller:
https://postimg.cc/MMYLb0J3

Photo of cables after taking out generic controller and plugging to Raspberry Pi. Photo is of back of Raspberry Pi. My Raspberry has soldered pins:

https://postimg.cc/GBCz7BQW

Photo of my Raspberry Pi Zero's front which looks different to the one in the guide: (in this photo the white cable got unsoldered, ignore that)
https://postimg.cc/KkQ9gZtd

Can you help me with the photos provided and guess what what might I have done wrong? A million thanks in advance. Thank you so much 💗💗

Rodrix33333, 2022-10-08:
Schematics of https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/ refer to Raspberry Pi 3. This is why I can't follow...

I think I have soldered the wrong pins. It would be great if you could confirm...🙏🙏🙏

Rodrix33333, 2022-10-08:
I am reading and reading to try to find the issue.

I read that that GPIO18 has an output of 3.3V and that the Leds work at 5V so I need to add a logic level converter. Is this true for this project? I am confused.

Rodrix33333, 2022-10-08:
I feel so dumb :/... It's just two wires and I can't get it right. I would appreciate some help.

I have reviewed everything so many times and I am not following...

The article posted on this GitHub has a diagram of Raspberry Pi 3 pinout.

I have a Raspberry Pi Zero WH. I searched on Google how to connect this to a 2812B led strip, and it is asking me to use a logic level converter from GPIO 18 to change the 3.3v to 5v. Some even add a resistance.

I am confused. I don't know what to follow and I have one last chance to not burn this thing again.

onlaj, 2022-10-08:
> Schematics of https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/ refer to Raspberry Pi 3. This is why I can't follow...

It doesn't matter, they follow the same standard.
Using your photo I marked which pins you should use: 

[image]

I assume that since you wrote 

> I searched on Google how to connect this to a 2812B led strip, and it is asking me to use a logic level converter from GPIO 18 to change the 3.3v to 5v. Some even add a resistance.

you probably already found this website: https://www.thegeekpub.com/15990/wiring-ws2812b-addressable-leds-to-the-raspbery-pi/
Third schematic shows a wiring that doesn't require any converters. You can try to follow it, but it seems you will have to strip your cables anyway so I would suggest you to just follow wiring from the site I linked earlier.

Rodrix33333, 2022-10-08:
Thanks so much! Now the wiring is much more clear.

I will finish debugging the software part and then come back to this ticket, before closing it, and report back. I hope this also helps other newbies like me :).

Rodrix33333, 2022-10-10:
Success! Thanks so much. This was hard.

I was able to run the software and connect the LEDs. Now the leds change color according to the BACKGROUND color setting.
However they do not light  up when pressing a certain key.

My assumption: there is an issue with the MIDI connection.

I have the following setup:

PIANO -> (Midi out  to Midi in cable) -> Scarlett 6i6 audio interface -> (Midi out cable to USB / OGT using Mio interface) -> Raspberry pi

How can I check that the device is actually receiving midi data?
Thanks so much in advance!

onlaj, 2022-10-11:
> How can I check that the Raspberry device is actually receiving midi data?

There is a test file for that: https://github.com/onlaj/Piano-LED-Visualizer/blob/master/tests/printmidimessages.py

The thing is that this interface does not work anymore for me either. It worked right after the release of version 1.4, so it's not likely the fault of the code changes. The interface also works without any problems with my PC. 
As a temporary solution, I connect both the piano and the RPi to the PC (via USB-USB interface), and configure the Synthesia program so that it sends the signal to the RPi.
I hope this is an issue only on my setup because of my defective RPi. So far no one else reported problems with no signals from piano.

Rodrix33333, 2022-10-11:
Thank you so much for your help!
How can you setup Synthesia to do that? What's the name of the
configuration or setting? Thanks again

Rodrix33333, 2022-10-12:
Some useful information:

After running for the first time on ssh, the visualizer file, now it boots everytime without me doing anything. It seems that there is a bug that only affects fresh installs and this is why you are not able to replicate.

I have connected the piano to the Raspberry pi directly and it's working! However now I lost the connection to the computer.

Can you point me to a page or name of the configuration of Synthesia to do that?

Thanks a ton!

onlaj, 2022-10-19:
> How can you setup Synthesia to do that? What's the name of the

Hey, sorry for late response.
You just have to enable "my notes" in output device in Synthesia settings.

Rodrix33333, 2022-10-21:
Thanks so much!!!