onlaj/Piano-LED-Visualizer #389: Web Interface not working

Rodrix33333, 2022-10-08:
I am now trying with the second and last Raspberry Pi Zero WH that I have.
I am trying to setup all the software part before I continue with the wiring which I haven't figured out yet (seprate issue).

I cannot get the web interface working.

Steps I did:
1. I flashed the SD with the latest 1.3 image.
2. The Raspberry Pi boots successfully and shows the menu.
3.  I created the wpa_supplicant.conf  and ssh files and put my wifi network
4. I was able to successfully find the Raspberry Pi at 192.168.0.8
5. I SSH'ed into it using pi@192.168.0.8 using password raspberry, SUCCESFULLY.
6. I attempt going to the Chrome browser to http://192.168.0.8/ and nothing happens (Connection rejected)
7. I go back to SSH and type sudo raspi-config. I enable SPI.
8.  I attempt going to the Chrome browser to http://192.168.0.8/ and nothing happens (Connection rejected)
9. I cd to /home/Piano-LED-Visualizer and run "sudo -E /usr/bin/python3 visualizer.py". The console never returns anything like if it is running somehting.
10. I attempt going to the Chrome browser to http://192.168.0.8/ and nothing happens (Connection rejected)
11. Control+C to stop the console... and I am stuck.  I have no idea how to proceed. I have tried everything in my reach...

Please help :.(

P.S: Trust me, I am not "lazy". I have read all the 46 issues and every single comment.
I have also watched all of the following videos and read every single comment. I just cannot get this to work. I hope you can help :/
https://www.youtube.com/watch?v=IZgYViHcXdM
https://www.youtube.com/watch?v=31xEZO2boNs
https://www.youtube.com/watch?v=lbLjS3s74fU
https://www.youtube.com/watch?v=s8uX3LVx21E

onlaj, 2022-10-08:
> I flashed the SD with the latest 1.3 image.

The latest is 1.4 which uses services as a method to run Visualizer. You run it with

`sudo service visualizer start` (restart/stop)

You can stop the service and then run Visualizer with standard `sudo python3 /home/Piano-LED-Visualizer/visualizer.py` command to see console output and errors.

Rodrix33333, 2022-10-08:
Thanks a ton!

Should I update the code via GitHub or is there a link to the 1.4 image? (I cannot find it).

Then I should do:
sudo service visualizer stop 
sudo python3 /home/Piano-LED-Visualizer/visualizer.py

Correct?

If I connect to SSH to IP 192.168.0.8, will the url to visit the web console be http://192.168.0.8/?

Thanks so much for your help!!! :)

onlaj, 2022-10-08:
https://github.com/onlaj/Piano-LED-Visualizer/releases/tag/v1.4

> Then I should do:
> sudo service visualizer stop
> sudo python3 /home/Piano-LED-Visualizer/visualizer.py
> 
> Correct?

Yep

Rodrix33333, 2022-10-08:
You are the best! I will test and report asap.

If I connect to SSH to IP 192.168.0.8, will the url to visit the web console be http://192.168.0.8/?

Will the IP be the same?

onlaj, 2022-10-08:
Yes, unless you specified other port. It should show an error if default 80 is already taken by some other app.

Rodrix33333, 2022-10-10:
Update:
Installing v1.4 with BalenaEtcher results in blank screen on LCD. I tired with two SD cards.
Reverting to original SD card with v1.3 results in correct loading of menu on LCD.

Rodrix33333, 2022-10-10:
Any commands I can run on v1.3?

At least that one is able to boot successfully and I can access SSH.

onlaj, 2022-10-10:
Try to kill all python processes:

`sudo pkill python3`

and then launch visualizer with

`sudo python3 /home/Piano-LED-Visualizer/visualizer.py`

Rodrix33333, 2022-10-10:
Thanks. 

I run that and I got:
`Inport set to rtpmidi raspberrypi:Network 128:0
playport set to rtpmidi raspberrypi:Network 128:0`

It does not output anything more.

http://192.168.0.8/ still gives ERR_CONNECTION_REFUSED.

Suggestions?
Thank you

onlaj, 2022-10-10:
Is there no message about webinterface starting? It seems that it doesn't load at all. Do you still have sd card with version 1.4? Try to load it, SSH into RPi and type
`sudo service visualizer stop`
and then, just to be sure
`sudo pkill python3`

then launch visualizer 
`sudo python3 /home/Piano-LED-Visualizer/visualizer.py`

Rodrix33333, 2022-10-10:
No there is no additional message, nor any message about webinterface starting.

[image]

The v1.4 does not boot at all. Blank screen. 

Suggestions?

onlaj, 2022-10-10:
> The v1.4 does not boot at all. Blank screen.

If the Visualizer script didn't start it will be blank, but it doesn't mean that you can't SSH into RPi.

Rodrix33333, 2022-10-10:
No luck :/

If I put in the SD with v1.4 I cannot even login to SSH.

onlaj, 2022-10-10:
That's weird. Did you configure wifi connection through the wpa_supplicant file?

Rodrix33333, 2022-10-10:
I re-did the whole procedure again with 1.4.

We are getting somewhere...
[image]

Finally! I see the web interface.
So I guess I have to manually launch the sudo python3 /home/Piano-LED-Visualizer/visualizer.py every time?

onlaj, 2022-10-10:
I'm not sure yet.
The reason I switched the boot process to service is because some users had problems with old method, which was cron. It was launching the script before RPi connected to the internet. 
With service method it waits for internet, but maybe something else is blocking the script.

You can try to add delay by editing it with
`sudo nano /lib/systemd/system/visualizer.service`

and then under `[Service]` and before `ExecStart=sudo python3 /home/Piano-LED-Visualizer/visualizer.py` add 
`ExecStartPre=/bin/sleep 30`

Rodrix33333, 2022-10-10:
Thanks so much! :)
For others, here is the summary:

- Use version  v1.4. You will see a blank screen
- Login into SSH
- Run on SSH:
- sudo pkill python3
- sudo python3 /home/Piano-LED-Visualizer/visualizer.py

Do this everytime you turn on the device.

closed 2022-10-10 by Rodrix33333