onlaj/Piano-LED-Visualizer #390: Velocity mode

Rodrix33333, 2022-10-12:
The velocity mode lights up correctly the key led; however it fails to correctly light up the side leds. They fade out very quickly and only one led remains lit.

I see the code is affected by control pedal. This might be the issue. How is the pedal supposed to affect this behavior?

What is the pedal "dead zone" in the cars?

I have a continuous pedal controller which outputs different values (half pedalling is supported). If I understand I might be able to help fix a bug.

Thanks for this awesome project. It's incredible 💫🍦

onlaj, 2022-10-24:
I think I overcomplicated that part. It should be easier to redesign Velocity mode to ignore pedal signal and then add "the pedal keeps the LEDs lit" applicable to all light modes.

Rodrix33333, 2022-10-27:
I think that's a great idea! I like the new branch proposal.

closed 2022-11-02 by onlaj