onlaj/Piano-LED-Visualizer #393: Scale mode bug

Rodrix33333, 2022-10-26:
Scale mode produces an index out of range,  when playing too many notes.
This is a bug.

onlaj, 2022-11-09:
Can you tell me how to reproduce that error? I thought it occurs when you press the highest or lowest key, but it works just fine for me.

Rodrix33333, 2022-11-10:
Playing many keys, fast, simultaneously, triggers the error. I was just
playing classical music, nothing too weird.

onlaj, 2022-11-10:
I was just pressing many notes randomly, but it doesn't crash for me. Can you show me the full error message?

Rodrix33333, 2022-11-10:
Change the scale to E minor (g major) I think that was it.

onlaj, 2022-11-15:
I checked all scales, still can't reproduce that error. Can you show me error message? At which line the error occurs?

closed 2025-02-22 by onlaj