onlaj/Piano-LED-Visualizer #394: Rainbow mode expansion (feature)

Rodrix33333, 2022-10-27:
Rainbow mode is amazing. 

To make it even better I suggest:

1) be able to change the gradients (so it's just a hue animation of reds, for example. Or a rainbow of fire colors).

Maybe what we actually need to do is to add a new functionality to gradients: time and scale for the animation.

2) to make the speed time factor, change depending on how many notes are pressed.

Rodrix33333, 2022-10-27:
Let me rewrite this:

1) add time and scale to gradient mode.
2) add the possiblity of multi point gradients.

That's all we need. No need to touch the rainbow mode.

Nbobito, 2022-10-28:
What if there was a way to create custom animations and gradients? They could be created using some tool in the web interface, then saved as an animation file.

Rodrix33333, 2022-10-28:
No need to have animation files, I believe. Multi point gradients would be enough. As a setting.

What type of animations do you have in mind. I am interested

onlaj, 2022-10-28:
>add the possiblity of multi point gradients.

That I can do. I think SeeMusic has this feature too, so it will work great together.

Rodrix33333, 2022-11-02:
Looking great!!! Do you think you can add the time animation feature (as in Rainbow)? Could be 0 for no animation. Looking forward to testing it!

Rodrix33333, 2022-11-08:
It's looking great! Could you add the time feature for the multicolor?

Rodrix33333, 2022-11-08:
> Nvm, I thought about it for a while and I can see some usecases. Added to
> to-do list.

Awesome!

rodrix385, 2023-04-27:
I am back!!! I am looking forward to expanding your source code. Are you still active? :)

onlaj, 2023-04-27:
Yes, but my RPi is fried and I still didn't buy new one, so I can't make any tests. I will try to find some replacement.