onlaj/Piano-LED-Visualizer #401: Connection setup for Sevilla's Soft MIDI USB-USB device

sreich, 2022-11-13:
Hello, I've been struggling trying to figure out how to connect this USB connector from Sevilla soft

Especially because both ends of the port are the male connector. The connector coming out of my piano has the printer USB on one end and on the other that goes to the PC normally would be USB A male connector

Which I cannot connect both of those together without a female to female adapter.

What is the proper connection for this? My goal is to connect my piano to it and my raspberry pi and my desktop

My understanding is that I plug it into a USB hub but then where do I plug that in and where would the piano get plugged into?

onlaj, 2022-11-14:
Hello.
Sevilla's device is used to connect RPi (through USB OTG hub) to your PC. Your piano should be connected to RPi through the same hub.

sreich, 2022-11-14:
What does the full connection string look like? The USB ports as I mentioned are unexpected and would require adapters it seems. What does your setup for it look like?

onlaj, 2022-11-14:
USB OTG hub connected to RPi.
Piano connected to USB OTG hub.
PC Connected to Sevilla's device.
Sevilla's device connected to USB OTG hub.

There is another setup I currently use that doesn't require Sevilla's device. Tomorrow I will add it in README file.

sreich, 2022-11-15:
Excellent thank you that helped and I got it working well

closed 2022-11-15 by sreich