onlaj/Piano-LED-Visualizer #402: Save Raspberry pi settings?

tutseryday, 2022-11-14:
Hello,

Is there any way to save settings or changes? For instance I have a smaller keyboard and so I have to change the LED count and offset. But whenever I turn off my Raspberry Pi, the settings don't save and I have to manually change the LED count and offset every single time I turn on my Pi. 

Any way to save settings? Thanks in advance

tutseryday, 2022-11-14:
> 

Sorry for not being clear. Yes it resets to default after each restart. Offset goes back to 0, LED count goes back to 176 after every restart

onlaj, 2022-11-14:
Changes should be saved into "settings.xml" file. Did you manually install Visualizer or did you use image from [releases](https://github.com/onlaj/Piano-LED-Visualizer/releases)?

If you are able to SSH into your RPi try this command:

`sudo chmod a+rwx -R /home/Piano-LED-Visualizer/`

It should allow Visualizer to make changes to files.

closed 2022-11-14 by tutseryday