onlaj/Piano-LED-Visualizer #403: Synthesia over RTP

ezocustoms, 2022-11-14:
Hello, first of all thank you for this great project.

I am having issues connecting Synthesia over RTP. I tried the desktop solution but kept getting disconnected. I am now on Android and am using Midi connector https://abrahamwisman.com/midiconnector. 

I have enabled lights in Synthesia as Finger-based channel, and set my Pi input to rtpmidi.

In the app I see both "raspberrypi" (port: 5004) AND "raspberrypi/RtMidiIn Client-RtMidi input" (port 55043). Also see "Midi Connector Free Virtual Port 1" from the app.

What should be the input and what should be the output? I have tried all configurations and none of which work. Anyone else have the solution to this?

ezocustoms, 2022-11-14:
@AbrahamWisman I see you are the creator of this program, thank you for creating it. Would you be able to provide any assistance?

AbrahamWisman, 2022-11-15:
Hi, 

1) The input should be Midi Connector Free Virtual Port 1.
2) In synthesia set the output as Midi Connector Free Virtual Port 1.
3) In Midi connector set the output as raspberrypi/RtMidiIn Client-RtMidi input. 

The RTP-MIDI deamon/service on the raspberry pi (should but) might not respond with a confirmation that is is connected. But that can be ignored. 

Hope this helps!

ezocustoms, 2022-11-15:
> Hi,
>
> 1. The input should be Midi Connector Free Virtual Port 1.
> 2. In synthesia set the output as Midi Connector Free Virtual Port 1.
> 3. In Midi connector set the output as raspberrypi/RtMidiIn [...]

Thank you for your reply! Unfortunately the keys don't light up and there seems to be no input from my Piano in Free play, it's as though it isn't detected in Synthesia.

AbrahamWisman, 2022-11-15:
Can you give a little more detailed information. Do the keys not light up at all? What kind of piano-synthesia connection are you using?

You want piano->synthesia->midi connector->virtualizer right?

1. Test the led demos on the virtualizer
2. Check with the "TEST" button in midi connector if random notes light up
3. Check if synthesia output is received in midi connector log and notes should light up

TIP: Sometimes rebooting stuff works wonders.

ezocustoms, 2022-11-15:
@AbrahamWisman 

Sure sorry, I'll try to explain better. The keys light up fine when I am using the visualizer itself. The visualizer itself works great. 

The problem arises when I switch the input to rtpmidi and try to connect to Synthesia. However when I switch the input on my Pi to rtpmidi, and try to connect it to Synthesia with your app midiconnector, nothing seems to happen. On midiconnector I have "Input" set to "Midi connector free virtual port 1", and "Output" set to "raspberrypi/RtMidiIn Client-RtMidi input". In Synthesia I have enabled "Midi connector 1" and have enabled enabled lights as Finger-based channel.

Yes this is what I want piano->synthesia->midi connector->virtualizer. As I am trying to learn pieces with synthesia, and I'd like for it to light up on the keys I have to press. However nothing lights up, and in the "Free Play" on Synthesia, nothing happens when I press any key on my Piano.

When I press "TEST", nothing on the Piano lights up either. I have rebooted the  PI, Piano and Android tablet.

AbrahamWisman, 2022-11-15:
Thank you. 

Can you test this please:
1. In free-play play some notes just by pressing the notes on screen, you should see them in the midi-connector log and notes should also light up.

2. What if you use windows or MAC with rtp connection to the pi?

ezocustoms, 2022-11-15:
@AbrahamWisman 

Thanks for the quick reply. I tried Windows rtpMIDI but my connection kept dropping, I always got "has not answered the connection request" message so I am trying with Android now.

When i first connect it to Midiconnector and hold the "TEST" button the key does light up. After I switch app to Synthesia and then go back to midiconnector, the "TEST" button no longer does anything and keys don't light up. I also get an error "RTP Error - {"name":"raspeberryipi\/RtMidiIN Client-RtMidi input","ip4":"myiphere","port":"35738","nwa":"wlan0"}.

Also what should my "Music output" settings be in Synthesia? I have turned off everything (My Notes, Background Instruments, Percussion, Metronome, Prevent local Notes") but Key Lights set to Finger-Based Channel.

ezocustoms, 2022-11-15:
Not sure what I did, changed some settings in Synthesia. But now when I press keys on my tablet in Free Play, they do light up on my Piano.

However, I still have no input from my Piano to Synthesia, as in if I press keys on my Piano, nothing happens in Synthesia. Also have that disconnect issue mentioned above, I keep having to reset the "Output" on midi connector to "raspberrypi/RtMidiIn" every time I get disconnected, which happens every minute or so.

ezocustoms, 2022-11-18:
Managed to get a stable connection. Keys light up in tutorial mode and in free play. But still no input from my piano to synthesia. Or midiconnect to Synthesia I guess, only output.

AbrahamWisman, 2022-11-20:
Have you set your piano as input in synthesia correctly?

As far as I know synthesia can detect USB devices directly. 
If you want bluetooth or rtp-midi you would need to route it via midi connector. 
So, if you want piano->midi connector->synthesia->midi connector->virtualizer you would need midi connector pro. That is because the free version can only handle 1 connection.

Let me know if you need more help.