onlaj/Piano-LED-Visualizer #406: No Piano input in Synthesia with Rtpmidi

tutseryday, 2022-11-16:
Hello,

I have managed to connect my Pi to Rtpmidi (Win 10) and connected it to Synthesia. If I go to Free Play and press notes on my computer, the piano keys light up. If I play a song tutorial, the lights on my piano also light up. However I have no input from my Piano to Synthesia, so I can't progress with the song and am always stuck at the beginning. If i press keys on my piano in Free Play, nothing happens either. Seems like only output is working, any idea on how to fix this?

Thank you

onlaj, 2022-11-18:
Hey.
What is your setup? Piano connected to RPi and then RPi (through RTP) to PC?
What are your Synthesia settings? Can you post screenshot from settings under "music devices" tab?

tutseryday, 2022-11-18:
> Hey. What is your setup? Piano connected to RPi and then RPi (through RTP) to PC? What are your Synthesia settings? Can you post screenshot from settings under "music devices" tab?

Thank you for the reply. Yes my Piano is connected to RPi, go the visualizer working and all.
Then I connect my RPi to my PC with RTPMidi successfully. The notes that should be pressed light up on my Piano, but no matter what I press on my Piano nothing happens, even in Free play.

[image]
[image]
[image]

onlaj, 2022-11-18:
Did you try pressing "connect ports" on Visualizer's screen or web interface?
Also, SSH into RPi and type `aconnect -i -l`, show us the command's output.

tutseryday, 2022-11-18:
> Did you try pressing "connect ports" on Visualizer's screen or web interface? Also, SSH into RPi and type `aconnect -i -l`, show us the command's output.

Wow that did it. I never tried "connect ports". Thank you so much for both this project and your assistance! Would love to support the project somehow

tutseryday, 2022-11-18:
Oops closed by mistake. That managed to fix the issue and I now have input on Synthesia, but the keys no longer light up for some reason

tutseryday, 2022-11-18:
> Try changing the input to your piano. Also, use web interface if possible, the GUI is more readable there. Under the "ports" tab you can find current port, secondary (with a button to easily switch between them) and list of port's connections.

Thank you I will try the web interface

closed 2022-12-15 by tutseryday