onlaj/Piano-LED-Visualizer #410: Pi now showing up on RTPmidi or midiconnector

NPDCenter, 2022-11-24:
Hello, I got the visualizer working good. I am trying to learn with Synthesia but no matter what I do, I cannot get my Pi to show up on either the desktop version (RTPMidi) or the Android version (midiconnector). I have tried everything from restarting rtpmidi,  refreshing ports, disconnecting ports, connecting ports. No matter what it just won't show up.

Any possible solutions? Thanks!

onlaj, 2022-11-25:
Hey.
Did you manually enter the IP of RPi in rtp midi app?

NPDCenter, 2022-11-25:
> Hey. Did you manually enter the IP of RPi in rtp midi app?

I haven't. I assume it would show up in the list but I will try that. On the Android version however, I don't see an option to manually enter IP and nothing shows up in RTP connections/Bonjour peers

closed 2025-02-23 by onlaj