onlaj/Piano-LED-Visualizer #413: LED continuously burning.

MilanScherpen, 2022-12-07:
Thank you for helping me with my previous questions so far, really appreciate it :). I currently have a malfunction in the LED strip. As a result, 1 LED lights up green continuously. Unfortunately, the function in the webface that cleans the LED strip does not work. The manufacturer of the LED strip says that I have to reset the LED strip with the corresponding remote control. However, I have demolished the infrared sensor for the connection to the raspberry. Is there a command that resets the LED strip? I have also tried updates, rebooted several times and reset settings to default function.

MilanScherpen, 2022-12-08:
> Hello.
> Is it first LED or some in the middle? Does unplugging and replugging the
> LED strip's power supply clear the LED?
> It sounds like the strip is most likely damaged and there is nothing you
> can do with it from a software standpoint.

I have already tried several times to turn the LED strip off and on.
unfortunately that makes no sense.  I also tested whether all other colors
within that LED are still working.  That is also the case.  Therefore it is
my assumption that the LED strip itself is not broken and that it is indeed
a software problem.  I understand from your response that there is
currently no function that software resets the LED strip?  (next to the one
in the web interface)

onlaj, 2022-12-08:
Pressing the middle key on Waveshare's screen also clears the LED strip. It sends a signal to turn off all LEDs, or to be precise, sets the colors of all LEDs to 0, 0, 0 (black). Moreover, replugging the power supply shouldn't turn that LED back on because when there is no animation running, the software isn't sending any new signals to the strip.

What is the exact behavior of the faulty LED? Does it turn green when you connect the power supply? Is it the first, last, or somewhere in the middle LED?

MilanScherpen, 2022-12-08:
The green light starts to burn as soon as power is applied to the LED strip. I have already checked whether the other colors still work within this LED. Blue and red both seem to work correctly, but the green one just never goes out. It is an LED that burns and is located about 20 LEDs before the end of the LED strip.

I also tried the middle button on the oled screen. It does indeed work for all other LEDs. I tested this by pressing keys and pressing that button at the same time.

onlaj, 2022-12-08:
So it's definitely faulty LED. The easiest way to fix that would be to just cut the damaged led and solder strip back together.

closed 2022-12-22 by onlaj

MilanScherpen, 2022-12-22:
Sorry for not reacting anymore. I will try that it seems the only option left.

onlaj, 2022-12-22:
Feel free to re-open the issue if needed :)