onlaj/Piano-LED-Visualizer #414: MY DGX670 can playback midi but input midi

MrVeryWhite, 2022-12-09:
my ES100 can work but dgx670 can‘t
how to cope it？thanks

MrVeryWhite, 2022-12-09:
Thank you for your reply, because I can't program.So I downloaded a zip file.Then use etcher install it into the card.
So I'm not very clear about the other operations you mentioned.

onlaj, 2022-12-09:
But you said ES100 (that's a piano) works, so I assume you already installed Visualizer and it worked correctly at some point?

MrVeryWhite, 2022-12-09:
> Hey. You really need to provide more informations. What exactly doesn't work? Does it work with Synthesia when connected directly to PC?

I can use mypie to play the media files inside, and then my electric piano can also play music.
But when I press the button, my led doesn't respond.
This means that my pie can output to my electric piano, but my electric piano can not output to my pie.

onlaj, 2022-12-09:
Try to change input, either using Visualizer's screen (if you have it) or web interface.

MrVeryWhite, 2022-12-09:
> But you said ES100 (that's a piano) works, so I assume you already installed Visualizer and it worked correctly at some point?
Yes, my es100 is available.But my gx670 doesn't work.

MrVeryWhite, 2022-12-09:
> Try to change input, either using Visualizer's screen (if you have it) or web interface.

It sounds like a technical job, and I may not be qualified for it.
Can you tell me why I downloaded the file plv1.3.zip in this.
But I installed it on my card A white screen is displayed when the computer is turned on with several pies.
i check home/ I didn't see the  piano-LED-VISUALIZER   
Am I missing a step?

I downloaded another image file from the website before, and after installing it, I can directly enter the Piano-led in the Raspberry Pi.

onlaj, 2022-12-09:
Does the screen shows Visualizer menu?

MrVeryWhite, 2022-12-09:
I tested PLV1.4 1.3 
Only plv1.0 SHOW THE VISUALIZER

onlaj, 2022-12-09:
Later versions requires internet connection, is your RPi connected to Wi-Fi?

MrVeryWhite, 2022-12-09:
> Does the screen shows Visualizer menu?
I think you may have forgotten to put the vISUALIZER in when you packed the image.

MrVeryWhite, 2022-12-09:
> Later versions requires internet connection, is your RPi connected to Wi-Fi?

my pie is zero no wifi

onlaj, 2022-12-09:
Then you'll have to use version 1.0. I might fix it in future versions so it doesn't need internet connection.

MrVeryWhite, 2022-12-09:
> Later versions requires internet connection, is your RPi connected to Wi-Fi?

i download PLV1.0  and install onto my SD card just now
It can be used. Thank you very much
Thank you for your reply. Thank you very much.