onlaj/Piano-LED-Visualizer #419: "Can't play this file" when trying to play Midi files.

ezocustoms, 2022-12-21:
It used to always work but recently, I am no longer able to play midi files, I always get the "Can't play this file" error. I'm trying to play the same files that worked in the past, and never had this issue until recently. I changed OTG because I needed more than 1 port, but everything else still works except for that. Not sure if that's the reason, because even with my new OTG it worked, but it no longer does.

Suggestions?

onlaj, 2022-12-21:
Hey, what about the "La campanella" included in repository?

ezocustoms, 2022-12-21:
> Hey, what about the "La campanella" included in repository?

Same error

onlaj, 2022-12-21:
Does learning mode work?

ezocustoms, 2022-12-21:
> Does learning mode work?

Yes it does

onlaj, 2022-12-21:
Can you SSH into your raspberry and run script manually?

`sudo service visualizer stop`

 and then

`sudo python3 /home/Piano-LED-Visualizer/visualizer.py`

then try to play midi file and check what error it throws

ezocustoms, 2022-12-21:
> Can you SSH into your raspberry and run script manually?
> 
> `sudo service visualizer stop`
> 
> and then
> 
> `sudo python3 /home/Piano-LED-Visualizer/visualizer.py`
> 
> then try to play midi file and check what error it throws

```
ERROR:webinterface:Exception on /api/change_setting [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/Piano-LED-Visualizer/webinterface/views_api.py", line 990, in change_setting
    return send_file(safe_join("../Songs/" + value), mimetype='application/x-csv', attachment_filename=value,
TypeError: send_file() got an unexpected keyword argument 'attachment_filename'
```

onlaj, 2022-12-21:
It looks like you are clicking "download song" instead of "play song".
Do you use the latest version of Visualizer?

ezocustoms, 2022-12-21:
> It looks like you are clicking "download song" instead of "play song". Do you use the latest version of Visualizer?

Yes I updated it and that's when it stopped working I think. I'm pressing the "Play" icon though

onlaj, 2022-12-21:
Did you update through webinterface/screen? There were some changes that could break something, the best way would be to just install the latest image https://github.com/onlaj/Piano-LED-Visualizer/releases/tag/v1.4

ezocustoms, 2022-12-21:
> Did you update through webinterface/screen? There were some changes that could break something, the best way would be to just install the latest image https://github.com/onlaj/Piano-LED-Visualizer/releases/tag/v1.4

Yeah I updated it from the webinterface. Do you mean formatting my microsd and reflashing it with latest image?

onlaj, 2022-12-21:
Yeah, I'm sorry about that but it's the easiest way.

ezocustoms, 2022-12-21:
> Yeah, I'm sorry about that but it's the easiest way.

No worries, thanks for your help!

closed 2022-12-21 by ezocustoms