onlaj/Piano-LED-Visualizer #421: can't connect to wifi

CriToX42, 2022-12-22:
Hello,
I have downloaded the image file and install it on my sd card using etcher, then I created an empty ssh file then I created the wpa_supplicant.conf file and replaced my wifi name and password. After this I eject the sd card from my laptop and put it in the rasberry, powered it. And... nothing but the activity led, white screen and no wifi connection. I searched the entire internet but i found nothing. Please Help I'm lost  (maybe drama queen)
Thanks
have a good day

onlaj, 2022-12-22:
Hello.
Did you install the latest (1.4) version?
Does your RPi have Wifi? There is a version without wifi and bt.
How long have you been waiting for it to start? First boot takes around 8 minutes, but it can take longer if your SD card is slow.
Also, double check if data in wpa_supplicant.conf is correct.

For a testing purpose: install clean Raspberry Pi OS and check if it also doesn't connect to Wi-Fi.

CriToX42, 2022-12-22:
Hello it's indeed the last version(1.4), my rpi have wifi (It's written W after the "zero" on it), I have made 2 other test with 20 minutes waiting, one with your version and one with Raspberry Pi OS lite without any better result.
Here is my wpa.supplicant.conf file:
```
country=IE #I'm wondering if i haven't to put fr there (already tried but maybe it's a assembly of things)
ctrl_interface=DIR=varrunwpa_supplicant GROUP=netdev 
update_config=1 
network={ 
scan_ssid=1 
ssid="sponso"
psk="scjmponso"
key_mgmt=WPA-PSK # on my phone it's written WPA2 PSK so maybe I have to put WPA2-PSK here?
}
```
Thank You

onlaj, 2022-12-22:
One other thing that comes to my mind: on what frequency does your wi-fi work? RPi has only 2.4GHz, so if your Wi-Fi is 5GHz it won't work.

CriToX42, 2022-12-22:
how can you find this info?

onlaj, 2022-12-22:
Are you able to connect your RPi to some monitor and keyboard? You could install full Raspberry OS to check if it is possible to connect to your wifi from graphical interface.
You can also try to make a hotspot from your phone and try to connect to it on RPi. This way we can find out if it's your router problem or RPi.

CriToX42, 2022-12-22:
I doesn't have any "small" hdmi and i haven't any mouse or keyboard with micro usb. 
But I may be able to find some in the week.

CriToX42, 2022-12-22:
i'm trying the hotspot technique

CriToX42, 2022-12-22:
Ok i tried but it isn't working better

CriToX42, 2022-12-22:
ok i'm dying, i just tried to remove everything exept the screen and the power supply: mini usb cable( no more led strip or external alimentation), I have formated my sd card and reinstall the image, then i put the sd card in the raspberry and... nothing but a white screen, I'm dying

closed 2022-12-22 by CriToX42

CriToX42, 2022-12-22:
Ok it works the solution for me has been to do a manual installation with the tool from raspberry, It's like etcher choose rasberry os lite then your sd card then open the setting (the cog wheel) and complete user and lan. After that access your raspberry by ssh and go through the manual installation guide. Thanks and sorry for my English