onlaj/Piano-LED-Visualizer #428: Getting Error in lib/functions.py

DemonJAZ, 2023-01-14:
Hi Everyone , Kudos to developers first of all for developing this, I am trying to get my piano setup , but my LED screen is completely white on boot, I am getting below error on running visualiser.py

/lib/functions.py , line 16 in get_ip_address s.connect(("8.8.8.8",80)) OSError: [Errno 101] Network is unreachable

Please help if you can

onlaj, 2023-01-14:
Hello.
Every version of PLV besides 1.0 requires an internet connection to start the script. Please connect your RPi to Wi-Fi, and it will start. 
Offline mode is a work in progress.
I'm sorry for your inconvenience.

DemonJAZ, 2023-01-14:
Hey @onlaj , thanks , I figured out that I need to connect to 2.4Ghz wifi rather than 5Ghz , it works perfectly , thanks man , you are a genius , Blessings to you

onlaj, 2023-01-14:
So it doesn't work on 5GHz too? Huh, I need to get my hands on the Raspberry Pi 4 and test it well.
I'm glad you were able to find solution.