onlaj/Piano-LED-Visualizer #430: My own syntesia

agustinvaquero, 2023-01-18:
Hi. Thank you for Your great work. 
I Am a C++ and OpenGL programer. 
Is posible, colaborate with you for develop an aplication like syntesia under Windows 10.
I would try capture the notes through the Web interface...  If It was share It with shared memory or send It by a socket. Thank you.

onlaj, 2023-01-18:
Hello
The Visualizer is already sending incoming notes through sockets when Midi Logging is enabled.
https://github.com/onlaj/Piano-LED-Visualizer/blob/master/visualizer.py#L302
https://github.com/onlaj/Piano-LED-Visualizer/blob/master/webinterface/__init__.py#L20

Instead of sending it to locahost:8765 you can send it to your app.
Good luck with your app!

agustinvaquero, 2023-01-18:
ok. thanks maybe I'll try it, if I have enougth time...

onlaj, 2025-01-07:
Closing due to inactivity. Feel free to reopen if you have more questions.