onlaj/Piano-LED-Visualizer #431: Led lights are not lighting up when keys are pressed

clancel, 2023-01-22:
Hi.  Since last night, led lights stopped lighting up when I pressed the key.  I checked wires and the animation setting and everything was fine. I haven't had this issue for a year so why is this happening.  Would you be to help me on this?  I look forward to your reply.

Regards,
Kim

onlaj, 2023-01-22:
Hello.
A few things I would check:
When you run the animation, is the LED strip lighting up at all?
Is your piano being detected under the ports list?
Try connecting your piano to another device, like your computer, to see if it's being detected there.

clancel, 2023-01-22:
Hi.  Thank you for your reply.  Yes, the led strip is lighting up when I run the animation.  Also, the piano is being detected under the ports list as well as my laptop when connected.

onlaj, 2023-01-22:
Do you use web interface? Under ports there should be "midi logging" section. After enabling it should display all messages from your piano. Does it show anything when you play?
There is also https://github.com/onlaj/Piano-LED-Visualizer/blob/master/tests/printmidimessages.py test file, you can use it to print all messages in console.
If you can SSH into you RPi, turn off Visualizer with
`sudo service visualizer stop` and then try to run it with
`sudo python3 /home/Piano-LED-Visualizer/tests/printmidimessages.py`

clancel, 2023-01-22:
Hi.  I initially used a terminal from my Mac OS to SSH into Pi.  But that was done a year ago and I haven't had to use it since.  Now that the I'm having an issue, I guess I need to use the terminal again.  If you can clarify what a web interface is and how I can locate the ports, that will be great. Thanks for your quick reply.

clancel, 2023-01-22:
Good news.  The lights are back on. I clicked update the visualizer and rebooted and the lights came back.  Hopefully, the lights will stay working.  Never knew that updating and rebooting would solve this issue. I now need to find out how to make the purple color more purple and orange lights more orange.  Currently, the purple looks more like pink and orange looks more like yellow.

clancel, 2023-01-22:
I forgot to ask you if it's necessary to update the visualizer. If yes, how often should you update it?

onlaj, 2023-01-22:
Only when I release [new image](https://github.com/onlaj/Piano-LED-Visualizer/releases) . I try to do it after introducing some major changes and fixes.

r0bvas, 2023-09-05:
Yes, newest version is very laggy. I had to clone to older version. How to check which version is on?  Thanks... so you guys know which one works for me.

onlaj, 2023-09-05:
@r0bvas in other thread you said that screen stays white. So it works on 1.4 but with white screen?
And what do you mean by laggy?

r0bvas, 2023-09-06:
ok, 2 things:

I have working version that is very stable, unfortunately I do not know how to check what version it is. It doesn't show on web interface.

I tried to flash new system with 1.4, but it wasn't working at all. I only got white screen. I was able to locate it via "Fing" software and it looked like it had an ip. But I couldn't connect it.
 
I also tried to "update " via web interface from my stable version  and that gave me "wi-fi" option on the web interface. But the MIDI was very laggy so , I couldn't use that one and had to  "clone" back to previous stable version that I still have .

onlaj, 2023-09-06:
You can always try manual installation. There is also [autoinstall script](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/autoinstall.sh) which does everything for you (not fully tested, errors can happen).

r0bvas, 2023-09-06:
No worries! take your time with 1.5. Thank you so much for all you do!

closed 2025-01-05 by onlaj