onlaj/Piano-LED-Visualizer #433: Recording Midi file

monkshoody, 2023-02-19:
Hi, I manually set up the visualizer and everything works fine... except the recording of Midi files. Everytime I try to record a Midi, I receive this error message:

> Traceback (most recent call last):
   File "/home/Piano-LED-Visualizer/visualizer.py", line 442, in <module>
     saving.add_control_change("control_change", 0, control, value, midiports.last_activity)
   File "/home/Piano-LED-Visualizer/lib/savemidi.py", line 46, in add_control_change
     self.messages_to_save["main"].append(["control_change", time_value, status, channel, control, value])
 AttributeError: 'SaveMIDI' object has no attribute 'messages_to_save'

I also tried to fix it myself by adding the `self.messages_to_save["main"] = []` in the `__init__` section of the class `SaveMIDI`. In this case the recording works for one record only. Afterwards I receive this error message:

> Traceback (most recent call last):
  File "/home/Piano-LED-Visualizer/visualizer.py", line 442, in <module>
    saving.add_control_change("control_change", 0, control, value, midiports.last_activity)
  File "/home/Piano-LED-Visualizer/lib/savemidi.py", line 46, in add_control_change
    self.messages_to_save["main"].append(["control_change", time_value, status, channel, control, value])
TypeError: list indices must be integers or slices, not str

Since I don't want to change the whole `savemidi.py` file, I stopped troubleshooting at this point. Has anybody detected a similar problem? Can anybody propose a solution?

Kind regards, and thanks for the great work.

onlaj, 2023-02-26:
Hey, sorry for the late response.

Check if the start_recording() method is being called by adding a print statement there, and then check the console to see if it's being displayed when you start recording.

If you're trying to declare self.messages_to_save under `__init__`, you have to declare it as a dictionary first:

```
self.messages_to_save = dict()
self.messages_to_save["main"] = []
```
Just to be sure, are you using the latest version of the code?

monkshoody, 2023-05-01:
Hey @onlaj,

Since the new update doesn't solve my problem about recording the Midi files, I manually debugged the code again. The problem is neither rooted in the `__init__` nor in the `start_recording()`-function, but in the `save()`-function of your `savemidi.py` file.

In the `start_recording()`, you initialise the `self.messages_to_save` like this:
```
self.messages_to_save = dict()
self.messages_to_save["main"] = []
```
But in the `save()`-function (line 71) you reinitialise it like:
```
self.messages_to_save = []
```
which means, that `messages_to_save` is no longer a dict, but an array. However, at this point you need to reinitialise it like:
```
self.messages_to_save["main"] = []
```

I manually updated my code and now everything works perfectly.

onlaj, 2023-05-01:
I used `self.messages_to_save = []` to only clear the content inside. I guess better way would be to just use `clear()`. It shouldn't matter tho because in the `start_recording()` it is reinitialised back to dictionary with `self.messages_to_save = dict()`.
I have no clue why it's not working for you.

monkshoody, 2023-05-02:
I see. Well, I don't know what's going on either. But now it works, so feel free to close the issue.
Great work by the way. I really love your project.

closed 2023-07-24 by onlaj