onlaj/Piano-LED-Visualizer #435: 3D Case model for the pi zero

fermaton, 2023-03-20:
Hello, thank you very much for all the tutorial and files.
When I open the link to the 3D case, I can only find a smaller version of the case, lacking the space for the DC socket connection and extra holes for the wires.
Can you please share the case that appears in the pictures of the tutorial?
Thank you

fermaton, 2023-03-20:
Great, thank you! I missed that.
Could be possible to create a version of the case without any holes at the top, in case I don't want to use the LCD hat? My intention is to use it in live concerts, selecting ready sequences for each song pressing a pedal

fermaton, 2023-03-25:
Hello again. I managed to modify the case to use the pi without the LCD hat. Tell me if you want it in case anyone asks for it too. Cheers

fermaton, 2023-03-31:
Ok! I made two versions: One of them also without the openings of sd card and hdmi port.

RPICase-nohat-stl:
https://drive.google.com/file/d/1J3K1zjFNvwhQPYBPKYtADZvWXB43Ihhq/view?usp=sharing

RPICase-nohdmi-nosd.stl:
https://drive.google.com/file/d/1po5jca9_TmGaduh1ZFR0zR5mhN0ucbju/view?usp=sharing

onlaj, 2023-03-31:
Thank you, I will add them to the repository.

> Could it be possible to add a shutdown button to safely turn off the PI when there is no LCD hat?

There is already such thing on [homepage](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/webinterface/templates/home.html#L350), and last time I checked it worked fine.

fermaton, 2023-03-31:
> There is already such thing on [homepage](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/webinterface/templates/home.html#L350), and last time I checked it worked fine.
Ok, thanks. I missed that!
I realized that with the sd card on, the case is too small filling the hole of the sd card. I have to lengthen the case to increase the space for the sd card. :(

onlaj, 2023-07-24:
I've added (your files)[https://github.com/onlaj/Piano-LED-Visualizer/tree/master/Docs] to repository, thank you!

closed 2023-07-24 by onlaj