onlaj/Piano-LED-Visualizer #439: No module named "websockets"

Escanor274, 2023-04-18:
Hi,
i´ve just finished the installation und wanted to start the pi, but there is now an error. Any idea how to fix it?
I copied the terminal:

sudo -E python visualizer.py
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 17, in <module>
    from webinterface import webinterface
  File "/home/pi/Piano-LED-Visualizer/webinterface/__init__.py", line 3, in <module>
    import websockets
ModuleNotFoundError: No module named 'websockets'

Sorry if its obvious for you, but i'm rly not a geeek und would be happy about some help :)

onlaj, 2023-04-18:
Run 
`sudo pip3 install -r requirements.txt`

Escanor274, 2023-04-23:
everything works fine now, thank very much. any way to give u a tip?

closed 2023-04-23 by Escanor274