onlaj/Piano-LED-Visualizer #440: Display remains white

Escanor274, 2023-05-01:
Hey, the visualizer worked absolutely fine at first, but now the screen doesn´t start, it is always white. It seems that only the screen has a problem because the leds react to the keyboard as they should. 
i have reinstalled the image several times now, but the screen is still not working. Unfortunately there is no error to see anywhere in the terminal.
any idea? :(

[image]

Escanor274, 2023-05-01:
i just have used test_screen and nothing happened. 
pi@raspberrypi:~ $ sudo python3 /home/Piano-LED-Visualizer/tests/test_screen.py
**********Init LCD**********
***draw line
***draw rectangle
***draw text
pi@raspberrypi:~ $

onlaj, 2023-05-01:
Hey,
I had this problem a few times, it fixed by itself after few hours.
You can also try to reattach the screen, maybe the pins are not connected correctly.

Escanor274, 2023-05-01:
that´s what i think too. i will buy a new screen and test it.
thank you for answering

Escanor274, 2023-05-06:
the old screen was indeed the problem. The new one works just as it should :)

closed 2023-05-06 by Escanor274