onlaj/Piano-LED-Visualizer #459: Support for 160 leds/m strip

David-Guow, 2023-07-14:
160 leds/m is more commonly available now in 2 m length. 
Would you please point me to a direction on which part of code needs to be modified to make 160 leds/m strip work?
Thanks

onlaj, 2023-07-14:
Hello, I'm working on a feature that will support any kind of leds density, stay tuned.
Or if you want to change it right now for yourself, take a look at [this commit](https://github.com/onlaj/Piano-LED-Visualizer/commit/668d594abc7209364da38425549b0731c92e3a4a)

David-Guow, 2023-07-14:
Thanks a lot for the fast reply, I will stay tuned for sure.
Tried to follow the steps from the commit and a few different density for my case. However the Led strip will not light up the right most 30 or so leds. I'm guessing it has a limit for leds set by the 144 /m strips? I'm not too good with coding, it's really appreciated to know you are working on this feature!

onlaj, 2023-07-14:
You have to change number of leds,it can be done through web interface, lcd screen's menu or by editing config/settings.xml
[image]

David-Guow, 2023-07-14:
I changed the led counts to 196 for my case.  The leds distances are "shrinked" and would not light up the parts corresponding the the right most 5 or 6 white keys.  Adjust the density didn't work for my case.

onlaj, 2023-07-14:
Yep, without editing the code it won't align properly with that led strip.
Basically you need to change 
`leds_per_meter = 144`
to 
`leds_per_meter = 160`

Then you can use LED Note Offset for final adjustments.

onlaj, 2023-07-18:
@David-Guow I updated the code. Now, instead of "low density" you can specify how many leds there are per meter on your led strip.

closed 2023-07-24 by onlaj