onlaj/Piano-LED-Visualizer #465: How does the "LED offset settings" work ??

AL1-1956, 2023-08-01:
Hello
I have a 176 LEDS/meter connected to the RPI and the second LED is perfectly aligned with the first note of my 88 keys keyboard
And every even LED is OK for the 70 first ones but then instead of the 72 one been lighted it's the 71 wich is lighted so there is now a mismatch.
I tried to modify the LED offset settings , but as i do not fully understand how it works .
So a little explanation will be greatly appreciated
is the offset about the LED number or the Key number ?
is it possible to input negative offsets ?
I tried  a few things but the results are quite odd (same LED for two different keys , LEDS not switching down when key released...)
Thank You and have a great day

Alain

onlaj, 2023-08-01:
Hello.
You can find explanation here: https://github.com/onlaj/Piano-LED-Visualizer/blob/master/Docs/features.md#led-notes-offset
It means that after the light number 55 the lights will be shifted by 1, for numbers bigger than 92 the shift will be equal to 2.
I will try to move it to somewhere more obvious and explain better.

AL1-1956, 2023-08-01:
Okay , so now I think I understand how it works.
I deleted the 2 default offsets and I saw that light 71 is no more lighted , wich is a good thing in my case.
I think that these offsets settings are a way to correct bad alignement wich occurs due to construction of the LED strip (by parts of 50 cms soldered together) 
I have to check LEDS after 72 but thanks for the answer and hope you will soon be able to make this PLV even greater
I was thinking of some kind of financial helping so that you can buy a new one...
But I do not have a PayPal account and I do not want to get one.
Alain

AL1-1956, 2023-08-01:
I forgot to tell that i used the OTG settings for the RPI and it is now seen as a midi device without need of the Sevilla gizmo that I will use for something else

closed 2025-01-05 by onlaj