onlaj/Piano-LED-Visualizer #466: Gradient mode in sequences crashes LED Visualizer

fermaton, 2023-08-22:
Hello.
I've just updated Led Visualizer manually. Everything is ok in sequences, thank you.
But when I add a new gradient mode in a step, it crashes the web interface and the Led Visualizer.
I tried to see anything strange in the xml file, but everything seems ok.
Thanks

onlaj, 2023-08-23:
Hello.
Thank you for reporting this bug.
I just [commited](https://github.com/onlaj/Piano-LED-Visualizer/commit/6ee3060353819694396b2588b91dc9aa298f130e) fix, can you check if it's working for you now?

fermaton, 2023-08-24:
Thanks. Now it works perfectly.
I've detected another strange behavior: If more than one step is set to "multicolor", when selecting the colors of the new step, the first one loses his previous colors and inherits the colors of the new one.
I was wondering too if there's a way to boot directly into the first sequence, without the need to access to the web interface and select it. Thanks for all the great work!

onlaj, 2023-08-24:
Nice catch! 
In fact, this occurs when there are two consecutive instances of the same color mode. I've fixed it. 
As for the loading sequence on boot, I'll think about it. I definitely want to avoid having it load by default, but introducing an additional option is a possibility. Maybe a checkbox "activate on boot" on sequence edit tab?
But first I need to tweak editing sequences. While I was working on fixing this bug I noticed how glitchy it can be.

fermaton, 2023-08-24:
Great! Thanks. That checkbox could be a nice addition for future versions, for live situations without hat.
But it works like a charm connecting through iPhone hotspot, anyway.

fermaton, 2023-08-26:
Hi! It seems with this new version, the "inherit colors problem" occurs with all the color modes, and it doesn't matter if they are consecutive or not. strange thing is in the xml everything seems ok. Tried with another device, and restarting the pi, but it keeps showing the same colors on different steps but same color modes

onlaj, 2023-08-26:
Did you try default example sequences? If it works fine with them, can you send me your sequences.xml file?
I just noticed there is bug with Fading and Velocity mode, the speed is not read correctly causing some weird problems. Maybe that's the reason? Try with light modes set to "Normal".

fermaton, 2023-09-02:
Hi! It seems with the new update everything works perfect with the sequences. I was using the example sequences. Using same color modes with different colors in different steps is ok now! Thank you

closed 2023-09-17 by onlaj