onlaj/Piano-LED-Visualizer #467: LCD Displyer remains white

ezocustoms, 2023-08-31:
**Describe the bug**
LCD Display remains white after flashing my SD card and inserting it into the Pi. The flash was successful.

**Installation Method**
I used the image PLV_v1.4. The flash was successful.

**Steps to Reproduce**
1. Flash image with Win32DiskImager to SD Card.
2. Create ssh and wpa file in boot partition.
3. Enter SD card into  Pi and turn on.
4. Wait indefinitely and screen remains white

**Error Messages**
Can't SSH into it, can't see the PI on router or FING.

**Expected behavior**
UI displays on PI after a certain time

r0bvas, 2023-09-05:
Same here, screen white etc. Stay away from 1.4

onlaj, 2023-09-05:
@ezocustoms apologies for the delayed response; I somehow missed that thread. 
Versions newer than 1.0 do not work without Wi-Fi. This issue will be resolved in version 1.5, which I plan to release later this month. Without checking the logs, I would suspect that your RPi is not connecting to Wi-Fi. 
Have you tried installing the base Raspbian Lite version and connecting it to Wi-Fi?

If it's possible, try connecting your RPi to an external display. This will allow you to check if the OS boots properly.

ezocustoms, 2023-09-06:
> @ezocustoms apologies for the delayed response; I somehow missed that thread. Versions newer than 1.0 do not work without Wi-Fi. This issue will be resolved in version 1.5, which I plan to release [...]

No worries! It should be connected to Wi-Fi, as it used to work and suddenly no longer does. I thought the PI/had/sd card were the issue so I got a new one but same issue persists. Though I tried 1.3 and it seems to work so I will be staying on that until the release of 1.5, which I look forward to! 

Thanks for this great project by the way

onlaj, 2023-09-17:
I released version 1.5, if you have any questions feel free to reopen this issue.
https://github.com/onlaj/Piano-LED-Visualizer/releases/tag/v1.5

closed 2023-09-17 by onlaj

r0bvas, 2023-09-17:
Super! Thank you 🙏 Robert Timeless Piano

https://github.com/onlaj/Piano-LED-Visualizer/releases/tag/v1.5