onlaj/Piano-LED-Visualizer #472: CPU Usage hovering 100% and input lag between key press and LED strip

ezocustoms, 2023-09-14:
**Describe the bug**
I used to have a Pi Zero 2 W with this project and it worked fine for a while, but my pi zero 2 w no longer works. I got a Pi Zero W instead because it was cheaper and the main git page for this project suggests a Pi Zero W anyway, but for some reason my CPU is almost always at 100% and there is very visible lag between key presses and LED strip lights.

**Installation Method**
Project image v1.3.

**Error Messages**
Using ``sudo python3 /home/Piano-LED-Visualizer/visualizer.py``

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/Piano-LED-Visualizer/webinterface/__init__.py", line 40, in start_server
    loop.run_until_complete(main())
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/Piano-LED-Visualizer/webinterface/__init__.py", line 35, in main
    async with websockets.serve(echo, "0.0.0.0", 8765):
  File "/usr/local/lib/python3.9/dist-packages/websockets/legacy/server.py", line 1070, in __aenter__
    return await self
  File "/usr/local/lib/python3.9/dist-packages/websockets/legacy/server.py", line 1088, in __await_impl__
    server = await self._create_server()
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1494, in create_server
    raise OSError(err.errno, 'error while attempting '
OSError: [Errno 98] error while attempting to bind on address ('0.0.0.0', 8765): address already in use
Unexpected exception occurred: [Errno 11] Resource temporarily unavailable
Traceback (most recent call last):
  File "/home/Piano-LED-Visualizer/visualizer.py", line 34, in singleton
    fcntl.flock(fh, fcntl.LOCK_EX | fcntl.LOCK_NB)
BlockingIOError: [Errno 11] Resource temporarily unavailable

**Environment**
Raspberry Pi Zero W

onlaj, 2023-09-14:
Hello.
It seems like there is more than one visualizer's script running.
Try killing all of them with `sudo pkill python3`.

ezocustoms, 2023-09-14:
> Hello. It seems like there is more than one visualizer's script running. Try killing all of them with `sudo pkill python3`.

Thanks. Still same issue with the 100% CPU

Though when running the command I know get this infinitely and rapidly

"Error: [Errno 2] No such file or directory: '/etc/hostapd/hostapd.conf'
Error checking hostapd package status
Error: [Errno 2] No such file or directory: '/etc/hostapd/hostapd.conf'
Error checking hostapd package status
Error: [Errno 2] No such file or directory: '/etc/hostapd/hostapd.conf'
Error checking hostapd package status
Error: [Errno 2] No such file or directory: '/etc/hostapd/hostapd.conf'
Error checking hostapd package status
Error: [Errno 2] No such file or directory: '/etc/hostapd/hostapd.conf'"

onlaj, 2023-09-14:
Oh, so you installed version 1.3 but then updated.
It's already fixed by @stephen322 https://github.com/onlaj/Piano-LED-Visualizer/pull/471 , I will merge it tomorrow.
Thank you for your patience.

ezocustoms, 2023-09-14:
> Oh, so you installed version 1.3 but then updated. It's already fixed by @stephen322 #471 , I will merge it tomorrow. Thank you for your patience.

Oh yes that is what I did. Thank you again

onlaj, 2023-09-15:
I merged the fix, please update your software and check if it's working correctly now.

ezocustoms, 2023-09-15:
> I merged the fix, please update your software and check if it's working correctly now.

This seems to have fixed the issue, thanks!

closed 2023-09-17 by onlaj