onlaj/Piano-LED-Visualizer #492: 144 leds/m strips

JayceeB1, 2023-12-01:
Hi!

Just a quick and simple question, 144 leds/m strips I've found are only one meter long, do you use 2 connected to each other?

Thanks for this awesome project, can't wait to build it :)

ArthaVerx, 2023-12-04:
You need it to be connected. try to find 2 meters or you can try to solder both 1m into one, then cut the last 28 led

onlaj, 2023-12-04:
Hi, you can also try to ask seller if they can solder it for you.

AL1-1956, 2023-12-06:
Hello 
You can find WS2812 in various length on AliExpress

https://fr.aliexpress.com/item/32476290984.html?spm=a2g0o.productlist.main.1.50fb3bfexnUZvG&algo_pvid=75d2d60b-31fb-4b4f-bf1b-dc7a2df36d89&algo_exp_id=75d2d60b-31fb-4b4f-bf1b-dc7a2df36d89-0&pdp_npi=4%40dis%21EUR%2131.94%2121.06%21%21%2133.67%21%21%40211b615317018906619347304e7b34%2112000034291084611%21sea%21FR%210%21AB&curPageLogUid=LmpI74ntyo3J

It's the French version but I think the link will work for you

Something to keep in mind is that the 2 meters strip is actually two 1 meter soldered together so the space between LEDs is at the soldering point different , so you need to adjust the settings

I think there is no ledstrip greater than 1 meter without a soldering point
 

Alain

JayceeB1, 2023-12-06:
Thanks Alain

It's perfect, I'm french too.

Thanks for the help guys!

mojeto, 2024-06-18:
Is 144 LEDs per meter a hard requirement? 
WS2812B 144 LEDs/m strip with 5050 casing is a too large to fit space on my piano. 
There are Ultra-Narrow LED strips with LED casing that should be hardware compatible otherwise e.g. [WS2812B 2020SMD](https://www.aliexpress.com/w/wholesale-WS2812B-2020SMD-200.html) Can this software work with those? (160 or 200 LEDs per meter)

onlaj, 2025-01-05:
@mojeto sorry for the (very) late response. Yes, the software allows you to change the density. You can use 60 LEDs per meter, 144, 160, 200 etc.

Closing due to inactivity. Feel free to reopen if you have more questions.

closed 2025-01-05 by onlaj