onlaj/Piano-LED-Visualizer #495: Request for sightreading practice

MiSTiC0, 2024-01-08:
**Is your feature request related to a problem? Please describe.**
I would like a feature that helps practicing sight reading.

**Describe the solution you'd like**
I used to practice on this website https://sightreading.training/

[image]

Something like that integrated with the LEDS to make it easier to identify the keys.

Could you implement at least for random keys? The rest of the options would be a nice addition too, but maybe a basic functionality would work just fine.

onlaj, 2024-01-08:
Hello.
You can already load any MIDI file and start learning from web interface.

[image]

MiSTiC0, 2024-01-08:
Was looking more for an auto-generate MIDI feature instead of having to look and load each time in order to practice exercises of random sheets on the fly, not learining sheets but practicing reading

onlaj, 2024-01-08:
I will think about it. 
I already had in mind adding some midi files with scales, chords etc. Unfortunately I lack knowledge about music theory and I don't know what should be included.

HangryPanda, 2024-05-12:
Here's a link to my google drive folder for Sheet Music and Midi files that includes the Major Scales, Chords, and Arpeggios https://drive.google.com/drive/folders/1DsPqHXGEBUJk0mb6meR66HLN1f_rdxxE?usp=sharing