onlaj/Piano-LED-Visualizer #499: Help with IPADTablet Setup

Sadvarsco, 2024-01-28:
Had anyone gotten this to work with an IPAD (or any external device)?
I made the LED strip and it works perfectly (I even have 2 144/M Led strips working to give extra effect), but I'm trying to use this with Synthesia via ipad or windows and I cant get it to connect/work regardless what I tried.

I worked through 1 the bluetooth tips, but couldnt get that to work,  I have the USB connector suggested, and still cannot get that to work with either. So I'm at a loss to have this LED strip to worth with Synthesia with ipad or windows. Can anyone point to me a "how to get it to work for dummies" guide?

onlaj, 2024-01-30:
Hello.
Do you have Sevilla's USB-USB device? That's the easiest way to connect RPi with pc or tablet.
If not, try RTP midi.
https://github.com/onlaj/Piano-LED-Visualizer/blob/master/Docs/features.md#learning-with-synthesia-

Sadvarsco, 2024-02-03:
I am using the Sevilla's USB-USB device, it seems like its not getting enough power.

I'll try out the links, thanks! My goal is to learn some piano by end of year.

closed 2025-01-05 by onlaj