onlaj/Piano-LED-Visualizer #506: led strip

PianoGuy1, 2024-04-15:
Hello, my issue is that only about 1 third of the LED strip lights up. 
I don't know if I did something wrong.

sreich, 2024-07-06:
have you tried setting in teh visualizer web, the length and offsets and playing with that? try using the animation section too, as it will hit all ends of it

if not, you'll want to make sure you have power and that all connections are working. if they're not, then the 1/3rd lighting up could also be a hardware connection issue. if your pi doesn't really connect to it well, the light may still get lit up to some degree, but it's more of a hardware fluke than a sw command

onlaj, 2025-01-07:
Closing due to inactivity. Feel free to reopen if you have more questions.

closed 2025-01-07 by onlaj