onlaj/Piano-LED-Visualizer #508: Visualizer UI gets stuck upon booting

SnaplingTurtle, 2024-05-16:
**Describe the bug**
Raspberry Pi Hat Screen UI gets "stuck" on main menu.

**Installation Method**
I used the V1.5.1 system image, I used Win32 Disk Etcher to put the image onto my SD card.

**Steps to Reproduce**
- Download V1.5.1 System Image
- Use preferred program to put system image onto the SD card
- Put SD card into Raspberry pi
- Power on Raspberry pi
- Raspberry pi _**WILL**_ boot into the "main menu", but none of the buttons nor "hat joystick" will work.

**Expected behavior**
Once booted the buttons and hat will control the menu, but they don't.

**Environment**
Raspberry Pi Zero V1.3 (Not 2W so using WIFI to try to find error messages isn't possible).
Using 32Gb Sandisk SD card.
Win 32 Desk Etcher.

**Additional Context**
Not sure what is causing this, I've left the Pi on for at least half an hour in case it's still loading but it's not clear to me whether it's a hardware issue or a software one. It should be noted that at this point I haven't soldered the LEDS/Power cable to the PI yet, but I don't think that should matter.

onlaj, 2024-05-19:
Hello.
Can you connect the RPi to a monitor to check the logs? I've never tested the setup on a Zero without Wi-Fi; there might be something causing the crash if the network isn't available.

r0bvas, 2024-09-20:
Hi, did you manage to solve this? I am stuck on booting , but it looks like the issue is password. I haven't changed the password initially and now my system is locked. Thx

onlaj, 2025-01-07:
Closing due to inactivity. Feel free to reopen if you have more questions.

closed 2025-01-07 by onlaj