onlaj/Piano-LED-Visualizer #512: [Bug] [regression] master fails at starting visualizer in colormaps[name] = gradient_to_cmaplut(gradients[name], gamma) TypeError: unhashable type: 'list'

sreich, 2024-07-06:
**Describe the bug**

issue seems to be:
```
Traceback (most recent call last):
File "/home/Piano-LED-Visualizer/lib/colormaps.py", line 110, in update_colormap
colormaps[name] = gradient_to_cmaplut(gradients[name], gamma)
TypeError: unhashable type: 'list'

```

----- 
```
pi@raspberrypi:/home/Piano-LED-Visualizer $ sudo python3 visualizer.py
[2024-07-06 13:42:41] INFO - abcmidi package is installed
[2024-07-06 13:42:41] INFO - Namespace(clear=False, display=None, fontdir=None, port=None, skipupdate=False, webinterface=None, rotatescreen='false', appmode='platform', leddriver='rpi_ws281x')
[2024-07-06 13:42:41] WARNING - Can't load settings file, restoring defaults
[2024-07-06 13:42:41] INFO - Inport set to rtpmidi raspberrypi:Network 128:0
[2024-07-06 13:42:41] INFO - Playport set to rtpmidi raspberrypi:Network 128:0
[2024-07-06 13:42:41] WARNING - module '_rpi_ws281x' has no attribute 'ws2811_set_custom_gamma_factor'
[2024-07-06 13:42:41] INFO - Failed to load LED strip.  Using emu driver.
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/cmocean_thermal.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/matplotlib_rainbow.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/matplotlib_terrain.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/agama_mist.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/cmocean_phase.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/cmasher_pepper.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/colorcet_c2_MYGBM.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/matplotlib_inferno.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/agama_circle.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/matplotlib_ocean.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/agama_earth.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/cmasher_bubblegum.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/matplotlib_plasma.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/hpluv_100_50.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/cmasher_gem.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/hpluv_100_70.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/matplotlib_viridis.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/cmasher_neon.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/cmasher_tropical.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/colorcet_c1_MRYBM.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] WARNING - Loading colormap datafile Colormaps/hsluv_100_50.sRGB.data failed: 'function' object has no attribute 'items'
[2024-07-06 13:42:41] ERROR - Unhandled Exception:
Traceback (most recent call last):
File "/home/Piano-LED-Visualizer/lib/colormaps.py", line 110, in update_colormap
colormaps[name] = gradient_to_cmaplut(gradients[name], gamma)
TypeError: unhashable type: 'list'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/Piano-LED-Visualizer/visualizer.py", line 122, in <module>
cmap.generate_colormaps(cmap.gradients, ledstrip.led_gamma)
File "/home/Piano-LED-Visualizer/lib/colormaps.py", line 120, in generate_colormaps
update_colormap(k, gamma)
File "/home/Piano-LED-Visualizer/lib/colormaps.py", line 113, in update_colormap
logger.warning(f"Loading colormap {k} failed: {e}")
NameError: name 'k' is not defined

```

**Installation Method**
Your image, then updated with apt over time

I am on origin/master, clean

There is a regression between master and v1.5.1 tag.

I bisected and narrowed it down to this:
```

pi@raspberrypi:/home/Piano-LED-Visualizer $ git bisect good
dd3074f659cd7dd2a9d2675f8306bf0a53ec3f7d is the first bad commit
commit dd3074f659cd7dd2a9d2675f8306bf0a53ec3f7d
Author: onlaj <onlyoneaw@gmail.com>
Date:   Thu Apr 18 18:18:27 2024 +0200

Added caching decorator to gradient_to_cmaplut function for better performance

```

onlaj, 2024-07-06:
Thanks, I will commit a temporary fix.

onlaj, 2024-07-06:
Done, feel free to reopen the issue if you encounter similar error.

closed 2024-07-06 by onlaj

stephen322, 2024-07-06:
I guess lru_cache stopped working with lists at some point?
Can revert the caching decorator commit -- caching not strictly needed.  I intended it to be loaded once at startup.  Only "Multicolor" would need updating, but even that is just one colormap which would probably be changing anyway.  Maybe the idea was for future real-time reloading?

For the bug in displaying exception: {k} can be changed to {name}

onlaj, 2024-07-06:
@stephen322 On my setup gradient_to_cmaplut() runs 70 times on boot, that's why I added cache. It was lazy and easier way to speed it up a little without digging why it has to run so many times.

stephen322, 2024-07-06:
gradient_to_cmaplut runs twice for each colormap -- once for the web preview, once for the ledstrip.
Either way, the caching errors out on my system too...

I suppose it might help in the scenario that Multicolor is used and changed in a cycle.  But I'm guessing its fast enough.