onlaj/Piano-LED-Visualizer #515: [Feature] velocity based gradient or brightness

sreich, 2024-08-03:
I would like a setting similar to "rainbow velocity". But the velocity the keys are pressed at would instead affect the brightness

Separately, an additional feature on top would be to set a gradient. Say, if I want it to go from red to purple to white, instead of being stuck with the rainbow velocity 

This feature is incredibly useful for being able to practice the force you apply to keys, in ways that I don't think the ear can. So, really awesome feature that's already there. These additional features would really help my practice for dynamics at piano

onlaj, 2024-08-03:
Hey. If you set the mode to "velocity" the brightness will depend on how fast/hard you press the keys.

sreich, 2024-08-03:
That is true thanks for reminding me, that has other side effects though that I don't prefer, Because then we can only choose between: 

1) normal light mode which stays on when the key is still pressed
2) velocity mode which you can only set to slow, so it still wants to fade away. Which ironically makes it harder because I'm trying to analyze my velocity visually, and that takes some time to pause. But the fade interferes with that, for me at least.

So, it's *almost* there for my uses, for the brightness feature request one, but not quite. I would want that mode to behave more like "normal". Perhaps a 4th light mode state for "velocity" light mode? "slow, fast, medium, no fade (normal)" ?

Just a suggestion, depending on how you want to structure this sort of thing so it's consistent

stephen322, 2024-08-03:
For the gradients, if you don't like any of the colormaps (I personally use colorcet_c1_MRYBM for Velocity Rainbow), you can play with the offset/scale/curve to use a custom subset of the colormap.  (Scale and Curve can go negative too).

There's also the option of using the "^Multicolor" colormap to use your own gradient generated by the settings in Multicolor.

And for an advanced option if you're comfortable with it, you can make your own colormap and place it in the Colormaps directory.  I can help describe the format if needed.