onlaj/Piano-LED-Visualizer #519: White led lights

PianoGuy1, 2024-10-24:
Hello, my LEDs do light but they only stay white. I can't switch colors or animations.

onlaj, 2024-10-28:
Hello,
Could you SSH into your RPi and turn off the visualizer? You can use the command:
`sudo service visualizer stop`. After that, disconnect the LED's power to turn off the lights, and run https://github.com/onlaj/Piano-LED-Visualizer/blob/master/tests/lightkeys.py to test if the strip lights up correctly. If it’s still faulty, the wiring is probably incorrect.

PianoGuy1, 2024-10-29:
Hello, I don't know how to SSH. Could you help me?

dk865, 2024-12-07:
You can use the command on a Linux/Windows Computer: `ssh plv@(ip address)`, for example, `ssh plv@111.111.1.111` The password is `visualizer`.

Also, to open a terminal in Linux, hold `ctrl`, `alt`, and press `t`. On Windows, hold the Windows button, press `r`, and type `cmd` and press enter.