onlaj/Piano-LED-Visualizer #523: Midi recording doesn't work

taherromdhane, 2024-11-12:
When I try to record midi, the application seemingly crashes and restarts. From what I understood, there's an issue with the midi input logic in the code, I played around with midi settings but it works 1 out of 5 times on a good day. 

**Installation Method**
Project image, latest version (v1.5.1 - Hotfix)

**Error Messages**
I have this in the logs, I'm actively investigating so will update with more findings

[2024-11-11 17:20:06] Starting webinterface
[2024-11-11 17:20:06] Running disable_ap.sh
[2024-11-11 17:21:31] 'NoneType' object has no attribute 'send'
[2024-11-11 17:21:39] Unhandled Exception: TypeError - list indices must be integers or slices, not str
[2024-11-11 17:21:48] Inport loaded and set to Digital Piano:Digital Piano MIDI 1 20:0
[2024-11-11 17:21:48] Can't load input port: RtMidiOut Client:RtMidi output 130:0
[2024-11-11 17:21:48] hostapd package is installed
[2024-11-11 17:21:50] Starting webinterface
[2024-11-11 17:21:50] Running disable_ap.sh
[2024-11-11 17:23:03] Unhandled Exception: TypeError - 'NoneType' object is not subscriptable
[2024-11-11 17:23:11] Inport loaded and set to Digital Piano:Digital Piano MIDI 1 20:0
[2024-11-11 17:23:11] Can't load input port: RtMidiOut Client:RtMidi output 130:0
[2024-11-11 17:23:12] hostapd package is installed
[2024-11-11 17:23:14] Starting webinterface
[2024-11-11 17:23:14] Running disable_ap.sh
[2024-11-11 17:26:44] Can't load input port: Digital Piano:Digital Piano MIDI 1 20:0
[2024-11-11 17:26:44] Can't load input port: RtMidiOut Client:RtMidi output 130:0
[2024-11-11 17:26:45] hostapd package is installed
[2024-11-11 17:26:48] Starting webinterface
[2024-11-11 17:26:48] Running disable_ap.sh
[2024-11-12 08:40:12] Error while getting ports 'NoneType' object has no attribute 'poll'

**Environment**
- Python version: 3.9.2

taherromdhane, 2024-11-12:
When debugging here's the stack trace of the error:
Exception has occurred: TypeError       (note: full exception trace is shown but execution is paused at: <module>)
'NoneType' object is not subscriptable
  File "D:\Work\Projects\piano led\Piano-LED-Visualizer\lib\savemidi.py", line 50, in add_control_change
    self.messages_to_save["main"].append(["control_change", time_value, status, channel, control, value])
  File "D:\Work\Projects\piano led\Piano-LED-Visualizer\visualizer.py", line 495, in <module> (Current frame)
    saving.add_control_change("control_change", 0, control, value, midiports.last_activity)
TypeError: 'NoneType' object is not subscriptable

jka-kan, 2024-11-18:
Does changing the order in `start_recording` help? Initialize `self.messages_to_save` before setting `self.is_recording = True`?
I also noticed that in line 73 `self.messages_to_save` is initialized as a list. Is this intentional or should it be a dict with key "main" always available in case of something wants to write to it even though it is not expected?

taherromdhane, 2024-12-04:
@jka-kan changing the order fixed the problem for me, I might open a pull request for that