onlaj/Piano-LED-Visualizer #525: MIDI Hub + Synthesia learning light method not working

Z3NT1, 2024-11-19:
I have tried to put the rtpmidi Port as input in Visualizer, my Roland as output and the finger-based one for learning lighting. The piano plays randomly in the lighting menu, but then the signal does not reach the lights for future keys. 

I have also tried to put my piano as input in MIDI Hub and link Virtual Port 1 as output. I have tried to put the virtual port as input and output in synthesia but this way even the random notes in the lighting menu don't play.

The lights do turn on when I play the notes, the only thing that doesn't work is the learning mode  

I think I'm missing something and I don't know what it could be. 

My piano is a Roland FP-30X. 

Thanks in advance..

Z3NT1, 2024-11-19:
@AbrahamWisman Could you help me with the problem I have with the sound? THANK YOU!!

dk865, 2024-12-07:
Okay, I know how to fix echos in Synthesia. 
The echo is caused from your piano playing a note, pushing it to Synthesia via MIDI, then Synthesia pushing it back to your piano. There are 2 ways to fix this:

1. No Synthesia Output (Good if you aren't doing a lesson):
Go into Synthesia Settings, and look for Outputs. Remove your Piano, and now Synthesia will not output to it, and your piano will be the only thing playing the sound.
2. No Piano Output (Good for Lesson, change once you're done with Synthesia):
Go to your piano function menu, and look for a section MIDI. Once in there, or if it's not there, look for something named Local. Turn this off. Now when you play your piano, it will push the note to Synthesia without playing a note, and Synthesia will push it back, playing the note. This also allows for more notes to be played from Synthesia that aren't being played on your keyboard, like when previewing a song, compared to the first method.

Note: If using the 2nd method, when your piano is unplugged from Synthesia, it will note play notes, unless you turn Local back on.

Also, sorry for the off-topic comment, but can you take a picture of your wiring (if you still have it) so I can do mine correctly?

dk865, 2024-12-07:
Oh, I'm sorry. I thought you were experiencing errors from Synthesia...