onlaj/Piano-LED-Visualizer #526: Experience troubleshooting rtpmidi + pipewire? Connection drops

sreich, 2024-11-21:
Has anyone had experience or issues with trying to get the midi signals across machines? I have my piano connected to my desktop for pianoteq, I then use rtpmidid to forward the MIDI signals to the raspberry pi which is connected to LED strip 

It works well, it's just the reliability that's the problem. 

I'm not sure how to trace or troubleshoot this, or where to start exactly. I don't know if it is induced by pipewire itself, or an issue with rtpmidid... Or A combination of this tool and the other softwares

It seems to happen after a few hours of it being connected, eventually the connection will somehow drop. Rtpmidid will report that it dropped because it didn't hear an ACK back

Then I have to reconnect (in pipewire UI, drag my midi connection to rtpmidid again, and reselect it in the visualizer input). This repeatedly happens and I'm puzzled by it. I was think maybe it was the raspberry pi WiFi going into standby, so I connected an Ethernet but experience the same thing.

Very frustrating because of how erratic it is. I attempted to go with a hardware solution (Seville soft USB midi multiplexer, but it broke and they no longer seem to sell it and I can't find anything else like that) 

Any suggestions or tips on where to proceed on this is appreciated. It probably isn't a visualizer issue itself, but figured this is the best place to start and maybe someone already knows, or maybe my conclusion can help someone else in the future...

onlaj, 2024-11-21:
Hello.
Visualizer is using the older version of rtpmidid. I'm waiting for a new release, once it's done, I will update the release image.

sreich, 2024-11-23:
Okay excellent, I saw you comment over on https://github.com/davidmoreno/rtpmidid/issues/128

Thanks for the quick response!

I'll look forward to those releases, perhaps I'll luck out and it'll fix this issue trivially :)

closed 2025-02-23 by onlaj