onlaj/Piano-LED-Visualizer #529: Shell Access from Web Interface

dk865, 2024-12-11:
**Is your feature request related to a problem? Please describe.**
Yes, after SSHing into the device and running apt upgrade, the password was changed, and is no longer visualizer.

**Describe the solution you'd like**
It would be really nice if you could implement a section, maybe even hidden to be protected from unknowledgeable people, that allows for shell access, or in laymen's terms, a Terminal button.

**Describe alternatives you've considered**
The only alternative I can think of is SSHing into it.

onlaj, 2025-01-07:
Hello.
I never paid much attention to the security of the Visualizer because it was intended to work only locally, not be open to the internet, and Linux's built-in security should protect it well enough. A terminal inside the web interface sounds like a cool feature, BUT I once accidentally stumbled upon a Visualizer instance online. SSH access was unavailable, so port 22 was probably not forwarded, but port 80 was. If the web interface had a built-in terminal, I could have taken control of someone's Raspberry Pi.
By the way, an update shouldn't change the password. Are you sure you didn't do anything else?

dk865, 2025-01-09:
I'm 100% sure I didn't do anything else... I'm not typically the guy to say something happens randomly, and never thought I would with Linux, but here we are haha. Thanks, I will now close the issue.

closed 2025-01-11 by dk865