onlaj/Piano-LED-Visualizer #535: Which LED strip should I buy?

martinformi, 2025-02-18:
The description might be a bit confusing. There are no LED strips with 144 diodes per meter with length larger than 1 metre. Amazon, AliExpress.

Amazon FR:

[image]

AliExpress:

[image]

Should I buy 2x 1meter and solder them together? Please give me a hint. I bought:

[image]

But simply it will not work because there the strip is made of smaller strips joined together so the offset is not the same across whole length:

[image]

onlaj, 2025-02-18:
[Users reported that LED strips bought on Amazon are shipped in one meter strips, in that case you would need to solder them. I bought mine on Aliexpress and it was 2 meters long strip in one piece](https://github.com/onlaj/Piano-LED-Visualizer?tab=readme-ov-file#faq)

As for the gap where the strips were soldered, the software can deal with that.

martinformi, 2025-02-18:
Hmm, in my case it is soldered every 50cm... so I would need to resolder it, so that there is no offset. This is not good, I will buy 2 another strips from 2 different suppliers, and maybe I will get the right one. How it can deal with it? It would use adjacent LED instead?

onlaj, 2025-02-18:
There is a "Led Note Offset" setting, allowing you to align the LEDs to the keys.

The-Torrent, 2025-05-10:
I have an APA102 LED strip that is 77 led/meter.

I can likely adjust the code to make this work right?

jbne, 2025-05-12:
I recently found some 2m - 5m 144LEDs/m WS2812b light strips on the BTF Lighting official site. Unfortunately due to tariffs they are extremely expensive to ship at the moment.

I don't think 144 LEDs/m strips more than 1m+ are manufactured, most places that sell 2m+ just solder together 1m strips together themselves but it's an alternative to those who don't want to mess around with soldering.