onlaj/Piano-LED-Visualizer #536: midi stream data loss

jinshidong, 2025-02-19:
**Describe the bug**
the light cannot work correctly since the midi stream data for key on and key off suffering heavy loss.(the CPU(about 50%) and memory of Raspberry Pi Zero WH is normal)
**Installation Method**
I reporduce the project and all function can work, however, the midi stream is not stable and some package will loss, the loss rate is above 10% for a roughly estimation. Here is my midi cable connection: **korg d1(an electronic piano) -> midi to usb cable -> usb to usb micro-b cable -> Raspberry Pi Zero WH**.  
I have test **korg d1(an electronic piano) -> midi to usb cable -> PC** and it works well 
the port information is  **CH345:CH345 MIDI 1 20:0**
**

> 22:40:29.517 note_on channel=0 note=95 velocity=71 time=0
> here should be a note_off message
> 22:40:30.467 note_on channel=0 note=95 velocity=56 time=0
> 22:40:30.583 note_off channel=0 note=95 velocity=64 time=0

**
**Steps to Reproduce**
copy the image to  Raspberry Pi , provide the power for led individual, and connect  electronic piano to Raspberry Pi as follow
korg d1(an electronic piano) -> midi to usb cable -> usb to usb micro-b cable -> Raspberry Pi Zero WH.  

**Expected behavior**
most midi message should be record

[image]

onlaj, 2025-02-19:
Hello.
Can you try to run this test file? 
https://github.com/onlaj/Piano-LED-Visualizer/blob/master/tests/printmidimessages.py

jinshidong, 2025-02-20:
Output：
**Traceback (most recent call last):
  File "printmidimessages.py", line 1, in <module>
    import mido
ImportError: No module named mido**
I think it‘s my venv setting problems, I will try it again when I have time

jinshidong, 2025-02-23:
I highly doubt that my e-piano 3.5mm sounds out is conflict with midi output, after more testing and fact checking, I will close this issue.

closed 2025-03-03 by jinshidong