onlaj/Piano-LED-Visualizer #539: README suggestions

ferent, 2025-03-01:
This is not an actual issue, just improvements to clarify parts of the README for users that are buying this kind of hardware for the first time.

onlaj, 2025-03-04:
Hi, thanks for the suggestions. 
I’ve fixed some parts of the README. Let me know if anything else needs attention. I didn’t follow all the instructions because some things don’t require explanation. For example, you need to enable SPI, but you don’t need to know what it is. If I tried to explain everything, the README would be twice as long. This is a DIY project, so unfortunately you have to do a bit of your own research if you want everything to come together.

ferent, 2025-03-04:
Taking a look at the new branch, thank you!

onlaj, 2025-03-04:
Oh right... I did commit to the wrong branch, just great.

ferent, 2025-03-04:
That is a good improvement! Merge it! 🙂🙂

closed 2025-03-07 by ferent