onlaj/Piano-LED-Visualizer #546: V1.6 Screen Menu does not appear

NoWayOut76, 2025-04-09:
When flashing a new image onto an SD Card of the 1.6 release, then the screen will not show the UI. I can access via hotspot and webinterface is working well. Just no screen UI on Waveshare LCD TFT 1,44'' 128x128px.
Flashing on same SD card 1.51 release works fine. 
Updating from 1.51 to 1.6 works fine. UI appears.

jbne, 2025-05-05:
I'm also seeing this, I flashed 1.6 onto a brand new SD card. The screen is completely white but the hotspot and web interface works.

onlaj, 2025-05-10:
@NoWayOut76 @jbne After updating the Visualizer via the web interface, restart the Visualizer first, followed by the Raspberry Pi. This should cause the SPI interface to activate automatically. It's a temporary workaround to avoid having to create a new system image.

NoWayOut76, 2025-05-20:
I can confirm that enabling SPI fixes the issue.

To make it work:
Install image on SD card as described in the manual
Boot the image. There will be no display but the hotspot will appear and you can connect and add the raspberry pi into your wifi.
Get the Raspberry Pi IP address from your router
Use SSH (e.g. putty) to log in to the Raspberry Pi. execute sudo raspi-config
Enable SPI in the interface options menu
Reboot